lewisxhe/AXP202X_Library #3: Schematic

clow57, 2019-08-30:
Apologies as this is a more general T-watch question. I am trying to understand the circuit for the AXP202. I know that LDO3 is for the main chip, and LDO2 is for the display, but I don't know what the other LDOs power. DO you guys have a full schematic?

Thanks
Chris

cyberman54, 2019-09-02:
I would need this, too, to develop my software.

cyberman54, 2019-09-02:
@clow57 LDO3 ist not main chip, but GPS chip. LDO2 is LORA chip. DCDC1 is OLED display.

as stated here: https://github.com/lewisxhe/TTGO-T-Beam

lewisxhe, 2019-09-03:
@clow57 
LDO3: s76g/s78g chip power supply [controllable, 3.3V voltage required]
LDO4: s76g/s78g GPS circuit power supply [controllable, 1.8V voltage required]

lewisxhe, 2020-06-20:
Close the problem here, if there are still problems, please reopen

closed 2020-06-20 by lewisxhe