lewisxhe/AXP202X_Library #5: Use function pointers for read/write to i2c bus

cyberman54, 2019-09-09:
On multitasking systems, like ESP32 with RTOS, with more than 1 device on i2c bus (e.g. TTGO T-Beam with display and AXP192) it is necessary to have a mutual exclusion between tasks that access the i2c bus. Usually this is done by using semaphore mutex. All I2c read/write functions must take/give the mutex semaphore.

To realize this with this library it would be useful to use function pointers for `_readByte()` and `_write(Byte)` functions, so that user can have own functions which implement mutexing.

Without usage of this library will cause race conditions while acessing the i2c bus if another task needs access to i2c bus, too.

lewisxhe, 2019-09-18:
You are right. Originally, this library was written for TTGO T-watch. There are multiple i2c devices in T-Watch. I used semaphores in another library for mutual exclusion, but it is not here. Is open source, how can you change it yourself?

cyberman54, 2019-09-30:
Meanwhile i saw you implemented this.
Thanks for implementing!

closed 2019-09-30 by cyberman54