lewisxhe/AXP202X_Library #15: AXP202 GPIO1 'HIGH' voltage value

lyusupov, 2019-11-19:
Hi Lewis!

I've tried first to set AXP_GPIO1 on HIGH in my (T-Watch) sketch to activate BOOT0 in S76G V1.6 - with no success.
Then I've detached LoRa+GPS board and substituted it with "Basic" one.
I've uploaded your _axp202_gpio_output.ino_ sketch and disabled LOW setting in it's loop() function, as follows:

```
void loop()
{
    //! GPIO0, GPIO1 allows output high and low
    axp.gpioWrite(AXP_GPIO_0, HIGH);
    axp.gpioWrite(AXP_GPIO_1, HIGH);
    delay(1000);
#if 0
    axp.gpioWrite(AXP_GPIO_0, LOW);
    axp.gpioWrite(AXP_GPIO_1, LOW);
    delay(1000);
#endif
}
```
I've measured voltage on both AXP_0 and AXP_1 pins of the "Basic" board and it showed me some value which was below 1 Volt. I would expect to see 3.3 Volts or close to this value.

Lewis, I am not familiar with AXP202 datasheet yet. Could you, please, look on what is happening wrong here?
Or, please, give me an advice of where should I look at.

    Best regards,
    Linar.

cyberman54, 2019-11-19:
@lyusupov 
Did you set the GPIO mode before, using 
`int AXP20X_Class::setGPIOMode(axp_gpio_t gpio, axp_gpio_mode_t mode)`
?

lyusupov, 2019-11-19:
No, I did not. 
I don't see that Lewis does this action in his example sketch.
And I've made a brief look over axp.gpioWrite() implementation and noticed that it makes access to a CONTROL register. So I assume that the mode change is a part of the axp.gpioWrite() operation.
This is not it?

cyberman54, 2019-11-19:
> So I assume that the mode change is a part of the axp.gpioWrite() operation.
> This is not it?

No, it's not. You need to select the mode.

lyusupov, 2019-11-19:
Ok, I'll try. Thanks for the tip!

lyusupov, 2019-11-19:
> e.g.
> axp.setGPIOMode(AXP_GPIO_1, AXP_IO_OUTPUT_HIGH_MODE);

Yes, this snippet did the job. Thank you!

I'll keep this ticket open for a while. The _axp202_gpio_output.ino_ demo sketch may need some improvements being made.

lewisxhe, 2019-11-22:
Sorry, I have seen it now, there is no notification of email in GITHUB, I have changed it. Thank you @cyberman54

closed 2019-11-22 by lewisxhe

lyusupov, 2019-11-22:
@lewisxhe
Lewis, it is likely that the notification did not come because you are out of the [Watchers list](https://github.com/lewisxhe/AXP202X_Library/watchers) of your own project.
GitHub does typically add the project's owner into "Watchers" automatically. I suspect you likely excluded yourself from the list somehow... 😄