lewisxhe/AXP202X_Library #18: AXP192 on T-Beam V1.0 charge current control settings seems not work

fafu, 2020-05-06:
When using the newly introduced charge current control settings to any setting (according to getChargeControl() is on AXP1XX_CHARGE_CUR_450MA) then the getBattDischargeCurrent is reporting it is not charging. Even setting setChargeControlCur(AXP1XX_CHARGE_CUR_450MA) to the previously read value with getChargeControl() seems to cause this issue.

cyberman54, 2020-05-09:
@lewisxhe I made PR to solve this.

cyberman54, 2020-05-09:
@fafu A bug in `setChargeControlCur()` inadvertantly sets the charging enable control bit to zero, whenever the method is called. This disables charging. Thus, after you called this method once, you get report "not charging". My PR fixes this.

lewisxhe, 2020-05-10:
Thanks @cyberman54

fafu, 2020-05-10:
Hi both,
tested the PR on my local T-Beam 1.0 (equipped with a AXP192) - I could verify setting the same value as the previous read value is successful. A second test with a max charge current of 100mA was also successfully. Using the max charge current of 1000mA worked - but I only saw a max charge current of ~400mA but maybe this originates in my setup. This issue seems to be solved. Thank you.

closed 2020-05-10 by fafu