lewisxhe/AXP202X_Library #25: AXP192 in TTGO T-Beam 1.0 setPowerOutPut function for DCDC3

Makhaos, 2020-06-25:
Hi there, first of all, thank you for this library :)

I am working with a TTGO T-Beam 1.0 and I followed their repo here: https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam to configure the board to go to sleep. I noticed that DCDC3 is commented out for powering it down.
You mentioned that the T-Watch uses DCDC3 as the esp32 to power the main chip, so probably is the same for the T-beam?
I am reading current consumption of 40mA like this and would much prefer to get the consumption they promised of around 0.9mA.
Do you know if I can safely power down DCDC3 and can I wake it up afterwards as well with esp_sleep_enable?
Below I have a snippet of the code:

`    setChgLEDMode(AXP20X_LED_OFF);
    // axp.setPowerOutPut(AXP192_DCDC3, AXP202_OFF);
    setPowerOutPut(AXP192_LDO2, AXP202_OFF);
    setPowerOutPut(AXP192_LDO3, AXP202_OFF);
    setPowerOutPut(AXP192_DCDC2, AXP202_OFF);
    setPowerOutPut(AXP192_EXTEN, AXP202_OFF);
    setPowerOutPut(AXP192_DCDC1, AXP202_OFF);
    
    delay(1000);
    esp_sleep_enable_ext0_wakeup(emergency_button,0); //1 = High, 0 = Low
    esp_sleep_enable_timer_wakeup(time_sleep);
    esp_deep_sleep_start();
    Serial.println("This will never be printed");`

Thank you

cyberman54, 2020-06-26:
DCDC3 powers the ESP32 cou on on T-Beam. Thus, if you switch off DCDC3, you cut off power of ESP32, what means the CPU ist not in sleep mode, but just off.

To save power, you need to select one of the sleep modes of ESP32
https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/#:~:text=ESP32%20Sleep%20mode%20is%20a,it%20to%20retain%20its%20data.
and carefully design your code to control ther peripherals.

Since this is not an issue of this library, i suggest to close this.

Makhaos, 2020-06-29:
Thank you for your input

closed 2020-06-29 by Makhaos

jangrewe, 2020-08-01:
I'm currently using the ESP32's deep sleep mode (and leave DCDC3 enabled), but i noticed that the AXP library also provides a `setSleep()` function, and the example for it does turn off the DCDC3, too. I'm wondering if this can be combined with the `setShutdownTime()`, `setTimeOutShutdown()` or any of the other related functions to delegate the wakeup from the ESP32 to the AXP?
According to my multimeter, i'm already at 0.8mA when in deep sleep, by the way. But lower is always better ;-)

PS: If anyone knows how to switch off the dim red LED when in deep sleep, i'd be very grateful! :-)

lewisxhe, 2020-08-04:
@jangrewe ESP32 cannot be woken up by AXP192. setSleep can only be woken up by clicking the PEK button. SetShutdownTime and setTimeOutShutdown are all just to set the time of the button.

jangrewe, 2020-08-04:
Ah, that makes sense. Thank you for clearing that up!