lewisxhe/AXP202X_Library #27: setChargeControlCur() and getBattPercentage() not working?

jangrewe, 2020-08-06:
Hi, thanks for fixing `enableChargeing()`, this seems to be fine now!

When implementing this, i discovered that e.g. `axp.setChargeControlCur(300)` results in `axp.getSettingChargeCurrent()` returning `1000`, and it always returns `1000` no matter what value you provide to `setChargeControlCur()`.

And i also noticed that `getBattPercentage()` always returns `-4` - i wouldn't know if this is correct, and if so, how to get a percentage from that?

It would be great if you could have a look at those two. :-)

lewisxhe, 2020-08-07:
When 192 is used, its optional parameters become these, and the percentage cannot be obtained from the register. 192 does not support obtaining the percentage. You can get the battery voltage and calculate the percentage, although this is not accurate
https://github.com/lewisxhe/AXP202X_Library/blob/0833e114dcabc88d0905f39eca3464c6094046dd/src/axp20x.h#L555

jangrewe, 2020-08-07:
Okay, thanks for the info about the percentage.
But setting the charge current seems to work, just not setting the right current according to the value?

lewisxhe, 2020-08-07:
Yes, for 192, it cannot directly set the current value, but needs to use the provided parameters to set

closed 2020-08-19 by lewisxhe