lewisxhe/AXP202X_Library #31: getBattChargeCurrent() not working ?

patrickpollet1966, 2020-08-31:
Dear All,

for a few hours i try to use getBattChargeCurrent() to read the actual charging current.
(In fact i want to monitor the charging state by reading the battery voltage and the current)

The issue is that getBattChargeCurrent() always return 0 but the battery is currently charging as the battery voltage increase (and an ampermeter shows me that current is flowing in the battery accordingly with the value i previously set by setChargeControlCur() ).

Did someone already have successful results with getBattChargeCurrent() ?
Did i miss something or is it a bug ?

Best regards.

Patrick Pollet

lewisxhe, 2020-09-01:
see here  
https://github.com/lewisxhe/AXP202X_Library/commit/e6f71c1856eb6dc3e893d2e32c3c2404007d388b#diff-3dfcca383d0a4827abb38cc896f08cc0

You must enable the adc register before you can read the adc data

patrickpollet1966, 2020-09-01:
Thanks a lot, it works well now.
It was my fault, i have missed this point concerning the activation of the ADCs.
I was leaded in mistake because the battery voltage was working without.

closed 2020-09-01 by patrickpollet1966