lewisxhe/AXP202X_Library #34: Battery Cutoff voltage?

Bwooce, 2020-12-31:
Great library, thanks. I've been trying to see what the low-voltage cutoff is, and how it works, without much success.

[Page 28 of the data sheet](http://igorx.irk.ru/datasheet/N9z7wouuCuHQm917GdSc.pdf) says there are some low voltage interrupts on the 202 at least, but none of those numbers on the sheet line up with the actual numbers in the code. So I'm clearly reading it wrong, correction welcome. There's also nothing about the voltage levels that these would fire.

It appears to cut off at 3.2v, which is fine. On the T-beam 1.1 it left a LED on though, but I'm still working through if it's possible to power this board down cleanly e.g. not mess up the bus and leave the AXP uncontactable.

Thanks!

lewisxhe, 2021-01-04:
Are you talking about low voltage interruption?

Bwooce, 2021-01-04:
Yes, sorry for the lack of clarity.

lewisxhe, 2021-01-05:
This feature is not in this library, I will add it after testing.

lewisxhe, 2021-01-06:
You can check the setting of low voltage interrupt in the axp_base example.

It has two levels, and the two levels correspond to different interrupts.

https://github.com/lewisxhe/AXP202X_Library/tree/master/examples/axp_base

Bwooce, 2021-01-06:
I'll give that a try, thanks! Very comprehensive. Just one typo - PowerDonwVoltage -> PowerDownVoltage ?

lewisxhe, 2021-01-06:
Thanks, this is indeed a typo

Bwooce, 2021-01-06:
Nice, I have it compiled and running. The cutoff voltage won't set below 3.4 (sensibly), so I set that and VWarning1 to reasonable values and I'll have to wait for the battery to discharge. Looking good so far.

cyberman54, 2021-01-06:
@Bwooce Meanwhile i also found my T-Beam powered off by AXP192 by itself due to low battery, but the blue LED of AXP192 remained on. Any idea how to configure the chip, that the LED is automatically switched off if the chip powers down?

lewisxhe, 2021-01-07:
LED lights are always on? Whether to insert USB? Otherwise, it will not be in the steady state!

cyberman54, 2021-01-07:
yes, blue led remains on while device is usb powered and battery is low, so that axp switched off power.

Bwooce, 2021-01-07:
Looks like it works (it powered off), although I didn't see the warning level yet (was asleep). Trying again now.

Yes @cyberman54 I experienced the same - on PowerDownVoltage being reached the LED turns on. I have not tried `axp.setChgLEDMode(AXP20X_LED_OFF);` at the Warning level to try and avoid that, have you?. Discharging the battery takes a while, but I will add this to my code and retest overnight.

@lewisxhe The TTGO T-Beam 1.1 appears to turn the blue charging LED on when it shuts down due to low voltage. No USB connected. I think this is hardware and not software, but I will try forcing it off before the PowerDownVoltage level is reached. Do you have any other suggestions? Leaving the LED on when the device has turned off due to low battery is not good behaviour (battery will continue to drain and be damaged).

Bwooce, 2021-01-10:
Okay so retesting is slightly challenging as I can get serial output but not charge easily (without doing horrible things to a USB cable, I guess). 

It does not look like the Level1 interrupt is firing. Is it supporting on the AX192 @lewisxhe 

I have  `axp.setVWarningLevel1(LOW_BAT);` where LOW_BAT is 3100. It's not firing the interrupt, which was set with `axp.enableIRQ(AXP202_PEK_SHORTPRESS_IRQ | AXP202_PEK_LONGPRESS_IRQ | APX202_APS_LOW_VOL_LEVEL1_IRQ, true);` and checked for with `if (axp.isLowVoltageLevel1IRQ()) {`

I've witness the low voltage power off working now, and the blue led is not on after it auto-off's at the low voltage level I set (so that's working!), so I'm going to wait a while and see if it turns on later (which would be odd).

Bwooce, 2021-01-10:
Okay my case was obscuring the LED and it was daylight. Can confirm blue  LED comes on every time on low voltage power off, even if the last command was to turn it off.

cyberman54, 2021-01-10:
Maybe LED logic is swapped in hardware of T-Beam, thus LED off means on and vice versa?

lewisxhe, 2021-01-11:
I use the axp_base example, the code is as follows, can enter the low voltage interrupt correctly. It can run normally.

```
#include <Wire.h>
#include <axp20x.h>

AXP20X_Class axp;
bool  axpIrq = 0;

const uint8_t i2c_sda = 21;
const uint8_t i2c_scl = 22;
const uint8_t axp_irq_pin = 35;

void setFlag(void)
{
    axpIrq = true;
}

void setup()
{
    Serial.begin(115200);
    Wire.begin(i2c_sda, i2c_scl);

    //! Use the Wire port
    int ret = axp.begin(Wire, AXP192_SLAVE_ADDRESS);
    // int ret = axp.begin(Wire);
    if (ret == AXP_FAIL) {
        Serial.println("AXP Power begin failed");
        while (1);
    }

    Serial.println();
    axp.setVWarningLevel1(3100);
    axp.setVWarningLevel2(3400);
    uint16_t level1 = axp.getVWarningLevel1();
    uint16_t level2 = axp.getVWarningLevel2();
    Serial.printf("getVWarningLevel1:%u mV \n", level1 );
    Serial.printf("getVWarningLevel2:%u mV \n", level2);
    Serial.printf("getPowerDonwVoltage:%u mV \n", axp.getPowerDownVoltage());
    axp.setPowerDownVoltage(2600);
    Serial.printf("getPowerDonwVoltage:%u mV \n", axp.getPowerDownVoltage());
    Serial.println();

    //! attachInterrupt to gpio 35
    pinMode(axp_irq_pin, INPUT_PULLUP);
    attachInterrupt(axp_irq_pin, setFlag, FALLING);
    axp.clearIRQ();

    //! enable all irq channel
    axp.enableIRQ(AXP202_ALL_IRQ, true);

    axp.setPowerOutPut(AXP202_DCDC3, AXP202_ON);
    axp.setPowerOutPut(AXP202_EXTEN, AXP202_ON);
    axp.setPowerOutPut(AXP202_LDO2,  AXP202_ON);
    axp.setPowerOutPut(AXP202_LDO4,  AXP202_ON);
    axp.setPowerOutPut(AXP202_DCDC2, AXP202_ON);
    axp.setLDO4Voltage(AXP202_LDO4_3300MV);
    axp.setLDO3Voltage(3500);
    axp.setPowerOutPut(AXP202_LDO3, AXP202_ON);

    axp.setChgLEDMode(AXP20X_LED_OFF);
}

void loop()
{
    if (axpIrq) {
        axpIrq = 0;
        axp.readIRQ();
        if (axp.isLowVoltageLevel1IRQ()) {
            Serial.printf("isLowVoltageLevel1IRQ\n");
            axp.setChgLEDMode(AXP20X_LED_BLINK_4HZ);
        }
        if (axp.isLowVoltageLevel2IRQ()) {
            Serial.printf("isLowVoltageLevel2IRQ\n");
            axp.setChgLEDMode(AXP20X_LED_BLINK_1HZ);
        }
        axp.clearIRQ();
    }
}

```

BuckfastBeekeeper, 2021-01-19:
Hello, i have about the same problem with my 1.1  
When i load the "axp_base example" in platformio, i get 18 errors 
class "AXP20X_Class" has no member "setVWarningLevel2"C/C++(135)
'class AXP20X_Class' has no member named 'setVWarningLevel2'

do i need another Librabry than this one ? 
lib_deps = lewisxhe/AXP202X_Library@^1.1.2

BuckfastBeekeeper, 2021-01-19:
in Arduino IDE  with axp20x.cpp and axp20x.h   in the same directory it works.
Can't change the Libs in Platformio (newbe there).

st:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698

getVWarningLevel1:3096 mV 
getVWarningLevel2:3399 mV 
getPowerDonwVoltage:2600 mV 
getPowerDonwVoltage:2600 mV 

Are these values correct?

BuckfastBeekeeper, 2021-01-19:
axp.setPowerOutPut(AXP202_DCDC3, AXP202_ON);
with the multimeter 3.3V pins are 1.8V  ?  with this script.  ??

lewisxhe, 2021-01-20:
In platformio, please copy the warehouse directly and connect to lib_deps,
Because I did not release the version, it is not the latest.
```
lib_deps = https://github.com/lewisxhe/AXP202X_Library.git
```

DC3 is the pad next to pin 21 in T-Beam., and I can get 3.3v voltage normally.

BuckfastBeekeeper, 2021-01-20:
i added axp.setDCDC1Voltage(3300); 
now the 3.3 next to pin 21 = 3.3V, also the 3.3 pin between the 2 GND pins is 3.3V
axp.setDCDC3Voltage(3300); did not work on these 3.3Pins. the voltage stayed 1.8V 
DCDC3 is the power for the MCU ?? as i read somewere? 

i thought that the problems with my board where equal as written above. 

This board did work OK for some months. Now the RED led goes out after some time when it is in 5 Min. sleep mode.
It does not come out of this sleep. Can it be that there is also a problem in the sleep with DCDC3? 
for this sleep mode i use  #include <esp_sleep.h>
Are there other methods with this power control? 

I learned al lot from your examples/

cyberman54, 2021-01-20:
DCDC3 on T-Beam 1.0/1.1 powers the ESP32 MCU.

lewisxhe, 2021-01-20:
Sorry, I was wrong!  DC3 is the power supply of ESP32, and DC1 is the power supply of OLED.  For DC1 I tested, setting the DC1 voltage can make it set to 3.3V.  For power consumption, you can check the sleep mode of ESP32 and adjust it according to the application.

lewisxhe, 2021-04-12:
The issue has been open for a long time, and there is no update, the issue will be closed

closed 2021-04-12 by lewisxhe