lewisxhe/AXP202X_Library #36: [Temp  Readings] AXP 192 on TBeam ESP32 Lora32 (T22_V1.1 20191212)

carlkid1499, 2021-02-25:
I am able to flash the axp_temp example to the board I have. My question is regarding the temperature readings from the temperature sensor.

Are the temperature readings internal (chip temperature) or are they external environment readings? The reason I am asking this because the printed readings in the serial output don't really make sense. I don't think the external environment is 175 C or the internal chip is 175 C as well. Any thoughts?

# Serial Monitor Readings
16:48:39.415 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:48:39.415 > configsip: 0, SPIWP:0xee
16:48:39.415 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:48:39.415 > mode:DIO, clock div:1
16:48:39.415 > load:0x3fff0018,len:4
16:48:39.415 > load:0x3fff001c,len:1044
16:48:39.415 > load:0x40078000,len:10124
16:48:39.415 > load:0x40080400,len:5856
16:48:39.415 > entry 0x400806a8
16:48:39.531 > 175.90*C
16:48:40.532 > 175.90*C
16:48:41.532 > 175.90*C
16:48:42.534 > 175.90*C
16:48:43.534 > 175.90*C
16:48:44.535 > 175.90*C
16:48:45.536 > 176.00*C
16:48:46.537 > 175.90*C

# Compile and Upload Output
Sketch uses 216064 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 14188 bytes (4%) of dynamic memory, leaving 280724 bytes for local variables. Maximum is 294912 bytes.
esptool.py v3.0-dev
Serial port COM7
Connecting......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 10:52:1c:66:40:44
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 10922.6 kbit/s)...
Hash of data verified.
Compressed 17120 bytes to 11166...
Writing at 0x00001000... (100 %)
Wrote 17120 bytes (11166 compressed) at 0x00001000 in 0.2 seconds (effective 866.8 kbit/s)...
Hash of data verified.
Compressed 216176 bytes to 113932...
Writing at 0x00010000... (14 %)
Writing at 0x00014000... (28 %)
Writing at 0x00018000... (42 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (71 %)
Writing at 0x00024000... (85 %)
Writing at 0x00028000... (100 %)
Wrote 216176 bytes (113932 compressed) at 0x00010000 in 1.6 seconds (effective 1063.6 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 3071.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

lewisxhe, 2021-02-26:
The battery temperature function is not used on TBeam, so it returns an invalid value

carlkid1499, 2021-02-26:
@lewisxhe So you are telling me that the AXP192 temp function is only for reading the battery temperature which is not supported on the TBeam?

So the
```c++
float temp = axp.getTemp();
    Serial.print(temp);
    Serial.println("*C");
```
in the ax_temp example reads battery temperature?

https://github.com/lewisxhe/AXP202X_Library/blob/master/examples/axp_temp/axp_temp.ino

lewisxhe, 2021-02-27:
Sorry, I misunderstood.

[axp_temp.ino](https://github.com/lewisxhe/AXP202X_Library/blob/master/examples/axp_temp/axp_temp.ino) works fine on my T-Beam.

Under normal circumstances, the internal temperature monitoring is turned on by default. Please add the following code to the sketch and try again!

```
axp.adc2Enable(AXP202_TEMP_MONITORING_ADC2,true); 
```

carlkid1499, 2021-03-02:
I placed it inside the setup function. No luck still reading the same ~170 degrees Celcius.
Odds are I might have messed up the config bits for the AXP causing the temp sensor to be offline.

```bash
void setup()
{
    Serial.begin(115200);

    Wire.begin(i2c_sda, i2c_scl);

    // Use AXP202 by default
    // int ret = axp.begin(Wire);

    // Uncomment will use AXP192
    int ret = axp.begin(Wire, AXP192_SLAVE_ADDRESS);

    if (ret == AXP_FAIL) {
        Serial.println("AXP Power begin failed");
        while (1);
    }
    axp.adc2Enable(AXP202_TEMP_MONITORING_ADC2,true); 

```

I also have done stuff like this:
```bash
// Lora2 Power
    axp.setPowerOutPut(AXP192_LDO2, AXP202_OFF);
    // GPS Power
    axp.setPowerOutPut(AXP192_LDO3, AXP202_OFF);
    // EPS32 Power ALWAYS ON
    axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
    // Need to findout what this one is for
    axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
    // OLED Pins + some other Power
    axp.setPowerOutPut(AXP192_DCDC1, AXP202_OFF);
    //  Set Charging LED ON
    axp.setChgLEDMode(AXP20X_LED_LOW_LEVEL);
```

But I don't see why that'd mess with the temp sensor.

carlkid1499, 2021-03-11:
@lewisxhe Update, I flashed the base example: https://github.com/lewisxhe/AXP202X_Library/blob/master/examples/axp_base/axp_base.ino

And then tried the temp example again.

Success: ```bash
8:33:43.744 -> 53.70*C
18:33:44.767 -> 54.00*C
18:33:45.775 -> 54.00*C
18:33:46.751 -> 53.80*C
18:33:47.773 -> 54.00*C
18:33:48.753 -> 53.90*C
18:33:49.780 -> 54.10*C
18:33:50.752 -> 54.30*C
18:33:51.776 -> 54.10*C
18:33:52.756 -> 54.00*C
18:33:53.763 -> 54.30*C
```

The temperature readings make sense again. :)

lewisxhe, 2021-04-12:
It looks normal, but the temperature is a bit high. Like broken

lewisxhe, 2021-04-12:
The issue has been open for a long time, and there is no update, the issuewill be closed

closed 2021-04-12 by lewisxhe