lewisxhe/AXP202X_Library #42: [E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0

yawe13, 2021-04-11:
Hi,
I`m on ESP32 Sim800 TT-Go SIM800C AXP192 20200609.
Sometimes i get the error below: 
```
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
chip id detect 0x0
AXP Power begin failed
```
1. Any hint or clue what's causing the issue & how to fix it? Without battery monitoring it's pretty risky to put the unit outside.. If i press PWR 2 seconds i can fix it temporarly, but question then would be how to fix it on the field without physical access?

cyberman54, 2021-04-11:
Check if there are any other components on same i2c bus. They may block the bus, if powered off. Also, check pull-up resistors on ic2 bus.

lewisxhe, 2021-04-12:
This is an I2C line failure

closed 2021-04-12 by lewisxhe