lewisxhe/AXP202X_Library #45: Arduino Library Registry

r41d, 2022-05-16:
Hello there :)

Have you considered submitting this library to the official Arduino Library Registry? (https://github.com/arduino/library-registry)

That way, this library could easily be installed in the Arduino IDE via the built-in Library Manager.

As far as I can tell everything in the library.properties file of this three library already looks great, the only thing that would be necessary would be the a PR on the registry repo.
The process seems to be rather quick.
If you're okay with it I could also do that PR.

lewisxhe, 2022-05-17:
I have submitted a request to the `library-registry`

closed 2022-05-17 by lewisxhe

r41d, 2022-05-17:
Awesome, thanks for responding quickly :)